jks-prv/Beagle_SDR_GPS #327: Offline update

mihaifireball, 2019-11-19:
How can I make an offline update of my KiwiSDR?

jks-prv, 2019-11-20:
Offline? Do you mean without an Internet connection? Or do you mean by hand (i.e. Linux shell commands) without the Kiwi server running?

mihaifireball, 2019-11-20:
Without connecting it to the internet.

jks-prv, 2019-11-20:
There is no easy way to do this.

You can always use an sd card to "re-flash" the software on the Kiwi. However the sd card usually contains a Kiwi software version that is not current. For example the card supplied with the Kiwi is programmed by the factory with v1.2 which is way behind our current version (v1.346 as of today). Re-flashing also does not preserve your previous configuration changes.

On the admin page there is a "backup" tab where you can write an sd card which will contain the current software version and configuration changes. But this does not help you in updating the software.

What's the situation in which the Kiwi has no access to the Internet? It would need it only for a short period of time if all you need to do is update.

mihaifireball, 2019-11-20:
The receiver is installed at the country side where I don't have any internet connection available... Not even 3G network (it is in the mountains 🙂)...I thought that there should be a way to clone the software from GitHub and after that to update it manually.Thanks!

jks-prv, 2019-11-20:
"Clone the software from Github". That implies that at some point you are located at a place with Internet. Why not just bring the Kiwi with you next time you are there and let it update automatically?

But it is certainly possible to do a manual clone of the Github repo and place it onto an sd card bring the card with you next time you visit the Kiwi. Then use Linux shell commands to shut down the Kiwi server, move the current Kiwi software directory aside and copy the new one from the sd card. Your configuration will not be effected because it is stored in a different directory from the software (this is intentional just for this reason).

But I cannot walk you through this process. Cloning from Github and writing the sd card is completely dependent on what kind of host computer you are using (WIndows, Mac OS, Linux). And then there are the Linux commands on the Kiwi Beagle necessary (mounting the sd card, copying from it etc.) Although some of this is contained in pour documentation. You need to have some sysadmin skills already to do this..

mihaifireball, 2019-11-20:
Perfect! This is what I was looking for. Thank you!

closed 2019-11-20 by mihaifireball

jks-prv, 2019-11-21:
When you login to the Kiwi Beagle as root you'll be in the /root directory. Then you want to type these shell aliases to shut the Kiwi server down: "cdp" "mst". Then "cd" to get back to /root. Now move the old software aside with "mv Beagle_SDR_GPS Beagle_SDR_GPS.old". Then copy the new Github clone from your sd card, which should also be called Beagle_SDR_GPS into /root. Now you can rebuild the new version: "cdp" "mi" "ku". "mi" is short for "make install". "ku" brings the Kiwi server up again.

The configuration files are in /root/kiwi.config so they will not be touched by simply replacing /root/Beagle_SDR_GPS.