jks-prv/Beagle_SDR_GPS #335: Request for generic iframe extension

OH1KK, 2019-12-29:
I'd like to see generic iframe extension on KiwiSDR. It would be an extension which does not actually do anyhing deep with KiwiSDR, just load a contents of remote webpage and show it content in  KiwiSDR gui.

Extension configuration page on KiwiSDR's admin could include items like

- Extension name: rotator
- Extension title: Rotator control
- Help description: Rotator control for ALA1530LN loop antenna
- iframe URL for content: http://www.mysite.org/rotator/
- Window width: 400px
- Window height: 400px

Then result would be something like this.

[image]

With this generic extension there would not no more need to alter actual kiwisdr code to make new extension.

OH1KK, 2019-12-29:
I am working with this. I wrote proof-of-concept extension based on antenna switch extension and it seems to work on my kiwisdr. But extension is quality to make it public yet.

OH1KK, 2019-12-29:
Not perfect, but works for me. https://github.com/OH1KK/KiwiSDR-iframe-extension

jks-prv, 2019-12-29:
Why did you add code in {main,ext}.cpp and kiwi.h to set “have_iframe_ext”? It never gets used anyplace.

“have_ant_switch_ext” exists only for use by the code that shows the icon in /status (and hence sdr.hu <http://sdr.hu/>) when the antenna switch extensions is installed.

jks-prv, 2019-12-29:
This extensions has just the “default" .cpp code. For extensions like this I need to figure out a way to load then automatically at Kiwi server start time so you don’t have to recompile the server. That is, you just put the extension .js/.css code in /root/extensions as usual and it gets discovered when the server starts up. I think this would be very useful.

Let me work on that..

OH1KK, 2019-12-29:
> Why did you add code in {main,ext}.cpp and kiwi.h to set “have_iframe_ext”? It never gets used anyplace. “have_ant_switch_ext” exists only for use by the code that shows the icon in /status (and hence sdr.hu <http://sdr.hu/>) when the antenna switch extensions is installed.

I just mimic and_switch extension. I did not realize that those are not need. I removed {main,ext}.cpp and kiwi.h modifications.

jks-prv, 2019-12-29:
Okay, fair enough. I should have had a comment in the code saying what it was for, lol.

Your idea is very good. People have been asking for a similar thing to add chat features on the main interface. Maybe this will work for them.

jks-prv, 2019-12-30:
This extension is very useful. So I think it makes sense to put it into the base Kiwi software rather than as a user-contributed extension that has to be downloaded.

What do you think?

OH1KK, 2019-12-30:
I Agree. Feel free to adopt extension code into KiwiSDR.

jks-prv, 2020-01-09:
Will be in v1.365

closed 2020-01-09 by jks-prv