jks-prv/Beagle_SDR_GPS #336: WSPR extension "Log decodes to syslog" doesn't seem to work

ofadam, 2019-12-30:
On the WSPR extension page in the admin interface, I have the "log decodes to syslog" turned OFF and I'm still getting them in the log. Is this user error or a bug?
[image]
[image]

jks-prv, 2019-12-31:
It should be working correctly. It has to do with the difference between what's recorded to syslog (filesystem) and what's displayed on the admin page log tab. Basically all printf()s in the code are recorded on the log tab, but only some of them are recorded to syslog. The printf() routines have a parameter so you can specify. So when the code is running in the background (as normal) all the messages, including possible debug messages, can be seen in the log, even though they are not filling up the filesystem via syslog.

If you have that switch set to 'No' you should not see WSPR syslog messages when doing an 'msl' (make short log) in the build directory ('cdp' to get there) using the admin console or SSH'd into the Beagle.

ofadam, 2019-12-31:
Ah, that makes sense – I didn't know that the two log locations were independent.

closed 2019-12-31 by ofadam