jks-prv/Beagle_SDR_GPS #356: More receivers to support IQ streams

iw2nke, 2020-01-18:
Could you take in account for a new mode to support more channels? One use case is based on IQ streams for the community looking at more bands with external support for decoding as using wsprdaemon or similars?
If the resources scale as done in the past, probably up to 12 channels should be supported.

jks-prv, 2020-01-18:
This analysis has already been done. 8 audio channels is the largest number that can be accommodated using a BBG/B. And 8 channels allow 2 waterfalls at most. If you want more you have to buy another Kiwi or upgrade the BBG/B with a BBAI to get 14 channels.

closed 2020-01-18 by jks-prv

iw2nke, 2020-01-19:
> This analysis has already been done. 8 audio channels is the largest number that can be accommodated using a BBG/B. And 8 channels allow 2 waterfalls at most. If you want more you have to buy another Kiwi or upgrade the BBG/B with a BBAI to get 14 channels.

Thank you for your fast reply. So, in case of upgrade to BBAI, it will show immediately more channels or I will need a sw upgrade?

jks-prv, 2020-01-19:
The software has been merged so that the current version runs on BBAI as well. It will offer the 14-channel audio option on the admin page if it detects a BBAI. There are no waterfalls in 14-channel mode, only audio FFTs.