jks-prv/Beagle_SDR_GPS #362: Request bookmark switch for callsign?

RightRudder, 2020-01-24:
May I request that in some future release that the capability to add a command line switch containing my callsign be added so that I don't always have to type it in on the top bar of the remotre KiwiSDR.  I have bookmarked many KiwiSDR's around the world and do like to respect the requests to log in with my callsign but also have set up bookmarks to maximize the screen for spectrum and waterfall.  It would be nice if I could be logged in with my callsign automagically when I use my bookmarks.

jks-prv, 2020-01-31:
When you visit a Kiwi site for the first time using a particular browser, and enter your call/name, the value is stored in a browser cookie associated with that Kiwi site. Subsequent connects will set the call/name field again from the cookie value. Do you have browser cookies disabled generally on your browser?

RightRudder, 2020-06-22:
I have a list of URL's for key receivers I like around the world that are saved in an email I sent to myself.  The reason being, I can quickly access them from any machine where I happen to be.  Cookies only work from a common machine but I move around a lot.  Most KiwiSDR owners request that you enter your callsign at the top of page, but the command line keys that I include in my URL's set up the display to maximize the spectrum and waterfall displays which makes the top bar disappear and then I am not able to enter my callsign.  If I could include a parameter in the URL to set my callsign as soon as I access the link it would be nice.  Not a big deal but I thought I'd just ask in case it was a simple thing to add.  Thanks for all your hard work on this wonderful system.  I use my Kiwi every day as a band monitor.

jks-prv, 2021-02-23:
Fixed as of v1.431. See: http://forum.kiwisdr.com/index.php?p=/discussion/2215/ios-no-your-name-or-callsign-field-fixed-in-v1-431

closed 2021-02-23 by jks-prv

RightRudder, 2021-02-23:
Thank you very much for the time you spent to work on this.  I really appreciate it!