jks-prv/Beagle_SDR_GPS #401: Feature request: decoders (extensions) as javascript code

dblas, 2020-05-27:
As I understand, decoders are at this time, some kind of c code embedded in the server.
The server thus delivers the signal decoded by the method invoked by the client side.
This method implies some load on the server side.
It would be more efficient for the server to make this computation on the browser side (javascript is very efficient and webassembly is near C performance).

Of course that implies that every client receives the full signal bandwidth in IQ.

What do you think?

db

jks-prv, 2020-05-27:
No. The extension implementer is completely free to chose how the code is partitioned between C/C++ code running on the server and Javascript code running on the browser. If you look carefully at our existing extensions you'll see there are a wide range of implementations.

For example, the FSK (and Navtex) decoders are completely written in Javascript and operate on the audio stream data already present in the browser. Look at the file Beagle_SDR_GPS/extensions/fsk/fsk.cpp It's just a skeleton that does almost nothing. This design choice was made primarily because the original source of the FSK algorithm was written in Javascript, so we kept it that way. And also because it only needed the real-mode audio data (i.e. IQ data not required).

Other extensions can only function on the server side because it would take too much effort to re-code them in Javascript (e.g. DRM). Also, the performance of JS is not always guaranteed. Plenty of people use older mobile devices (or older browsers) with marginal JS performance.

And, as an extreme case, sometimes even the Beagle and browser aren't sufficient. All the computational processing for the TDoA extension is handled on a single x86 VPS server that also hosts kiwisdr.com. Only the user interface is handled in JS for TDoA.

If you're suggesting a move towards always transferring full IQ data and moving code to the client side then this project starts to become no different than all the other "USB IQ generator" plus "PC client program" SDR solutions. You'd be better off buying an HF+ Discovery and running SDR#. That's not what KiwiSDR is about.

dblas, 2020-05-27:
OK, I understand. Thank you to explain how it works. So Javascript is possible.
Something to notice: It's possible to block some decoder on the client side for performance reasons.
I used to code some decoders in javascript a few years ago using primary editions of WebCL and it was very efficient. Alas WebCL didn't make it.
Have a nice day,
db

closed 2020-05-28 by dblas

jks-prv, 2020-05-28:
A Kiwi admin/owner can disable extensions individually if they decide they don't want them to be used for whatever reason (performance impact or excessive use). For example some Kiwis have DRM disabled because of its performance impact on the Beagle (it also locks-out the use of other extensions on the remaining channels).

Many users put a Kiwi in IQ mode, then use PC "virtual audio cable" (VAC) software to direct the IQ stream to 3rd party applications that have decoder/demodulator modes the Kiwi doesn't. Kiwi audio streams (IQ and real-mode) can be recorded to a file using the "kiwiclient/kiwirecorder" python application (see our Github). There is a python app called "kiwinc" that is supposed to stream IQ data netcat-style from the Linux command line, but it was never fully finished. That could potentially be used with JS running under NodeJS or whatever.

Other than that you pretty much have to write your own Kiwi extension to be able to access streaming IQ data in JA on the browser in order to do something with it. Unfortunately there isn't any documentation on the API. You have to look at the existing extensions for examples of what to do.

dblas, 2020-05-28:
I know the VAC but not very useful in my case.

> Other than that you pretty much have to write your own Kiwi extension to be able to access streaming IQ ...
Yes, no problem but I do not have a KiwiSDR to test (I'm not interested in having one since I live in a city) but I'm very interested in using remote ones.

db

jks-prv, 2020-05-28:
Okay. Sounds like "kiwinc" needs to get working so you could stream IQ real-time to JS running under NodeJS (I'm assuming). I don't think there was much more to be done to get it working. I just got pulled away by higher priority problems.

But if you need to run JS on the browser because you want to display something there or do something to the audio (as opposed to offline processing) then writing an extension is really the only way to go.

dblas, 2020-05-28:
I think you're right, it would need a specific extension since native IQ is limited to 11 k it seems.
Decoding WFM with data will need far more bandwidth.
There's many things to do I think: APRS, technical encoding on classical modulations (FSK, BPSK, etc) like thermal probes, MOTOBRO, etc. All at a fingertip without needing any other tools than a browser.

db