jks-prv/Beagle_SDR_GPS #406: Noise Test Pulse weakly present even when 'OFF'

ringof, 2020-05-31:
Try here: [W1EUJ KiwiSDR](http://kiwisdr.w1euj.com/?f=1891.20cwz11)

Under the Audio Tab, with Noise OFF and Filter OFF, there is a weak periodic click in the audio with similar timing to that of the noise blanker test pulses. 

Pulses are even visible with WF set to Max 20 dB and Min -155 dB

jks-prv, 2020-05-31:
He just has an electric fence charger nearby with ~1 Hz discharge timing.

This is not a bug in the noise blanker test pulse code for a couple of reasons. If it were it would apply with even amplitude across the entire 0 - 30 MHz spectrum. If you enable the test pulse with the amplitude set to -65 dB the two pulse sources (fence and test) are of similar strength but with slightly different timing. You can hear the phase between them vary.

jks-prv, 2020-05-31:
For an interesting look at the innards of a modern fence controller see Dave Jones here: https://www.youtube.com/watch?v=2LuWr25xgzE

ringof, 2020-05-31:
I now see and understand - thanks for the rapid response.

Oh dear, another conversation with the neighbor...

closed 2020-05-31 by ringof