jks-prv/Beagle_SDR_GPS #408: Clicking pulse after enabling noise blanker

ofadam, 2020-06-07:
After enabling one of the new noise blanker settings, there's a noise pulse every second that you can both hear and see on the waterfall. This persists even after turning off the noise blanker - but disappears if reconnecting the browser to the Kiwi interface (not rebooting the Kiwi.)

jks-prv, 2020-06-07:
In the extension panel (click on noise blanker "more" button) you have the test menu set to something other than "test off".

ofadam, 2020-06-07:
I tried that to no avail - here are the steps I took to get the pulse (and reversing them doesn't make it disappear):

1. Go to radio.ofadam.com:8073
2. On Audio tab, select "Wild" for NB.
3. On "More" options, select "Post filter (Wild)."
4. Pulse should appear.

ofadam, 2020-06-07:
Maybe I didn't realize this... but both the initial NB selector and the "more" selector need to be the same?

Even so, I still get the waterfall pulse on "Wild" (although I don't think I can hear an audio pulse, so maybe it's just a visual annoyance.)

jks-prv, 2020-06-08:
All three of the items in that menu relate to the test mode. So "post filter (Wild)" means the test mode is on. I'm going to reword the menu a bit to make this more clear.

closed 2020-06-13 by jks-prv