jks-prv/Beagle_SDR_GPS #409: Cross-compilation error

daniel5458h, 2020-06-09:
Hello,

When trying to cross-compile, I reach the point of doing a make install on the Beagle. At this point, I get the following error:

root@kiwisdr:~/Beagle_SDR_GPS# make install
make[1]: Entering directory '/root/Beagle_SDR_GPS'
clang++-6.0 -Dv1_395 @../build/gen/Makefile.inc -DEDATA_EMBED -c -o ../build/obj_O3/web_embed.o web/web.cpp
In file included from web/web.cpp:20:
In file included from ./kiwi.h:22:
./types.h:4:10: fatal error: 'stdlib.h' file not found
#include <stdlib.h>
         ^~~~~~~~~~
1 error generated.
Makefile:737: recipe for target '../build/obj_O3/web_embed.o' failed
make[1]: *** [../build/obj_O3/web_embed.o] Error 1
make[1]: Leaving directory '/root/Beagle_SDR_GPS'
Makefile:902: recipe for target 'install' failed
make: *** [install] Error 2
root@kiwisdr:~/Beagle_SDR_GPS#

I think there are issues with ../build/gen/Makefile.inc, as it contains references to the host computer.

Thanks,
Daniel.

jks-prv, 2020-06-09:
It's "make install_xc" to go along with the "make xc" you (hopefully) used earlier. Not "make install". See file: CROSS_COMPILE

jks-prv, 2020-06-10:
You can't do a "make xc; make install_xc" on a non-Beagle machine (a proper cross-compile) followed by just a "make install" on the Beagle. You have to do both a "make; make install" on the Beagle. Otherwise the ../build/gen/Makefile.inc file will not be regenerated by the "make" properly. The CROSS_COMPILE file doesn't say that under step 4 of the setup instructions, so I will fix that.

When you do the "make" on the Beagle almost none of the .cpp files will be recompiled because your "make install_xc" on the non-Beagle machine will have copied over all the object files to ../build/obj* which should be current relative to your sources (may depend on if you're copying them by setting KIWI_XC_COPY_SOURCES to true or using a shared filesystem to keep the sources synchronized).

daniel5458h, 2020-06-10:
I was doing cross-compiling. So I did two things wrong. 

First, I mounted the beagle filesystem as RW using sshfs because otherwise I couldn't compile. I didn't guess the proper way to do it was to copy the entire tree to the host computer, so that did mess-up the ./build/gen/Makefile.inc. I suggest you state that you have to copy the tree on the CROSS_COMPILE file too.

Second, after doing a make install_xc on the host, I did execute a make install on the Beagle in order to create the compressed html/css/js files, as make install_xc did nothing on them. May I ask what should be the procedure or command  to recreate the mini files and their compressed versions on the host computer?

Thanks for you help,
Daniel.

jks-prv, 2020-06-11:
Don't copy the tree from the Beagle. Do a proper checkout of the project from Github on your development machine.

The minified/compressed files are only created when you do a "make install" or "make install_xc" (when cross-compiling) **on the development machine**. This is never done on the Beagle under any circumstances.

daniel5458h, 2020-06-12:
Thanks John, I'll give it a try!

closed 2020-06-12 by daniel5458h