jks-prv/Beagle_SDR_GPS #410: web/Makefile issues

daniel5458h, 2020-06-09:
Hello,

I'm trying to customize several things in the code by editing web/kiwi/kiwi.js. However, changes are not updated even after a full make clean / make / make install. I've deleted the web/kiwisdr.min.*, but make install complains now that he doesn't know how to regenerate the compressed files. 

Thanks,
Daniel.

jks-prv, 2020-06-09:
Something is really wrong with your build environment. Full builds from clean distros are done hundreds of times per day worldwide with no problems. Try "make clean_dist" ("mcd" alias) instead of "make clean", although this shouldn't make any difference.

Like your previous issue, what the output from the "dog" command/alias?

jks-prv, 2020-06-09:
Could you email the /root/build.log file after the mcd / m / mi (aliases) process to support@kiwisdr.com ? Thanks.

jks-prv, 2020-06-10:
On the Kiwi Beagle the Makefiles are only designed to let you recompile/reinstall as part of the software install/update process. Any other development process, such as the generation of minified files etc., cannot be done on the Beagle. This is because minification is done using a 3rd party site on the Internet and I don't want there to be any possibility of this accidentally   occurring from the thousands of Kiwis worldwide. So the Makefiles only allow this if the "make install" is run on a "development" machine, i.e. non-Kiwi Beagle.

So use a laptop or desktop of some sort to develop and move the files over when a "make install" has been run on the dev sys which will create the minified files. You could also use a network shared filesystem like NFS that is read-only on the Kiwi side.

daniel5458h, 2020-06-10:
Thanks John,

In my case, make was complaining because it didn't know how to generate the compressed files. I guess the problem was that the web/Makefile was not saying how to do so. Here:

web/kiwisdr.min.js: $(FILES_EMBED_JS_PKG)
	cat $(FILES_EMBED_JS_F2) >web/kiwisdr.min.js
	gzip --fast --keep --force web/kiwisdr.min.js

web/kiwisdr.min.css: $(FILES_EMBED_CSS_PKG)
	cat $(FILES_EMBED_CSS_KEEP_MIN) `$(FILE_OPTIM) -t7 -u -css $(FILES_EMBED_CSS_F)` >web/kiwisdr.min.css
	gzip --fast --keep --force web/kiwisdr.min.css

So, it says how to create the css and js main minifiles, but not the compressed ones. Of course, the compressed ones will be created as long as the minifiles have to be created, but that was not triggered in my case. Anyway, I manage to patch it for me.

Best regards,
Daniel.

jks-prv, 2020-06-10:
The minification process is very, very complex. The $(FILE_OPTIM) program takes into account things like if the file is small enough (generally less than one nominal network packet) then no compression is performed because it isn't worth the time in the browser to decompress. Also some combination of files are concatenated and some are not because of various limitations (e.g. the extension files are kept separate because they are dynamically loaded on-demand).

There are no explicit rules for most of the compressed files because their dependencies are dynamic. There is no make rule that could be written to express them.

daniel5458h, 2020-06-10:
Ok, for my case, what I have to do to make it work was to change these statements on the makefile by this:

web/kiwisdr.min.css.gz: $(FILES_EMBED_CSS_PKG)
	cat $(FILES_EMBED_CSS_KEEP_MIN) `$(FILE_OPTIM) -t7 -u -css $(FILES_EMBED_CSS_F)` >web/kiwisdr.min.css
	gzip --fast --keep --force web/kiwisdr.min.css

web/kiwisdr.min.js.gz: $(FILES_EMBED_JS_PKG)
	cat $(FILES_EMBED_JS_F2) >web/kiwisdr.min.js
	gzip --fast --keep --force web/kiwisdr.min.js

So essentially I tell the makefile how to create the gz file instead of the uncompressed one. Certainly it has to be a way to tell make to create both of them properly, but I'm no programmer...

Also, I think that any update on the non-mini, human-readable files should trigger the minification process too. Or maybe there is a way to do it manually, but I did not find it. 

Thanks.

closed 2020-06-12 by daniel5458h