jks-prv/Beagle_SDR_GPS #429: Feature Request: Digimode-Decoders for APRS-Packet, FT4, FT8 ..., PSK31

hubertmeier, 2020-09-22:
Hi,

the fork of openwebrx from jketterl has made a lot of improvements and new features which are still missing in the kiwi ... the whole bunch of digimode decoders would be very nice to have in the kiwi. 

Thank you

jks-prv, 2020-09-23:
Most of those digimode decoders simply call WSJT-X directly on the PC host. This is impossible on the Kiwi because that strategy can't be done in a way that doesn't disturb the Kiwi's real-time processing requirements on the Beagle. This is one of the trade-offs of the Kiwi not requiring a dedicated PC (i.e. there is no free lunch).

The internal WSPR decoder was created by very carefully extracting the WSPR code from WSJT-X and rewriting it to work in the Kiwi real-time environment. This was a long and difficult process. It is not likely to be repeated for ARPS, FT4, FT8, PSK31 et. al.

closed 2020-09-23 by jks-prv

lazywalker, 2020-09-24:
@hubertmeier have a look at https://github.com/lazywalker/DigiSkimmer if you're looking for a ft8/ft4 skimmer

jks-prv, 2020-09-24:
Thank you Michael. I had forgotten about this (I seem to be forgetting a lot of things lately).
I have added DigiSkimmer to the "listening/operating tools" section of the kiwisdr.com website.