jks-prv/Beagle_SDR_GPS #441: 'Squelch' feature in KiwiSDR?

MetroSon88, 2020-10-24:
hi all!

I am new to github so i am not sure if this is a place for feature requests or strictly issues.

I noticed on the University of Twente's WebSDR site, they have a nice "squelch" option like this:
[image]

However, on a KiwiSDR, such as: http://ka7u.no-ip.org:8073/?f=16180am I do not see such option:
[image]

Is there something I am not understanding?

jks-prv, 2020-10-24:
KiwiSDR has squelch on the NBFM mode, but no others. The fact that WebSDR has it doesn't influence us in any way. This feature has been requested many times, but there are no plans at present to add it.

narrowfail, 2020-10-30:
Hello,
why isn't in the current dev plans? Is it hard to implement? Just asking since I would be nice to have a squelch on SSB.

jks-prv, 2020-11-01:
Non-NBFM (i.e. SSB) squelch is difficult. There have been several detailed threads about the topic over the years on the Kiwi forum: http://valentfx.com/vanilla/discussion/1421/syllabic-voice-squelch  Basically, if I can't find a chunk of open source C-language code that solves the problem, and is also easy to adapt to the Kiwi software, it isn't going to happen. There is too much else going on.

hcab14, 2020-11-04:
@narrowfail @MetroSon88: kiwirecorder.py _does_ have a squelch feature, see 
https://github.com/jks-prv/kiwiclient/tree/jks-v0.1

Have a look at the options of kiwirecorder.py:
```
    -T THRESH, --squelch-threshold=THRESH
                        Squelch threshold, in dB.
    --squelch-tail=SQUELCH_TAIL
                        Time for which the squelch remains open after the
                        signal is below threshold.
```
A new wav file is generated each time the signal is over the squelch threshold.

jks-prv, 2020-11-04:
Hmm, I forgot about that (although I've never tried it). How well does it work to simply use a threshold above the median value of a history buffer? I assume things like VAD were invented because simpler schemes didn't work well for low S/N situations.

I guess I should try implementing the ring buffer scheme and see. I can use the same squelch UI control for setting the threshold (in the "audio" tab) as for NBFM, with an additional enabling toggle button.

hcab14, 2020-11-04:
> Hmm, I forgot about that (although I've never tried it). How well does it work to simply use a threshold above the median value of a history buffer? I assume things like VAD were invented because simpler schemes didn't work well for low S/N situations.

It did work well for me for digital burst signals (using either 10-15db thresholds). I have not tried it for speech-like signals and don't know how well it works for low S/N.

> I guess I should try implementing the ring buffer scheme and see. I can use the same squelch UI control for setting the threshold (in the "audio" tab) as for NBFM, with an additional enabling toggle button.

Besides the threshold there is another parameter which determines how long the squelch remains open after the signal has gone below the threshold, but that can perhaps be a fixed constant for now.

jks-prv, 2020-11-04:
After the reorganization of the noise blanker/filter controls there is now plenty of empty space on the audio tab. So a second slider could be displayed when mode != NBFM.

There are going to be substantial UI questions though. The "principle of least astonishment" probably says that the slider values should be independent for the modes = NBFM and != NBFM (and stored in separate cookies). And there needs to be a new visible indication that the squelch is open so that people don't complain that suddenly "there is no audio anymore" when they've just forgotten they had the squelch set from last use (maybe the mute icon can have a third state in yellow for squelch open).

And of course people will want the squelch parameters to be on the same level of importance as mode and passband. So it needs to be settable in the URL, dx tag parameters and elsewhere. So this quickly becomes the familiar story: the actual algorithm implementation is 10-20% of the work with UI and testing being the rest! lol

jks-prv, 2020-11-07:
Fixed in v1.419

closed 2020-11-07 by jks-prv