jks-prv/Beagle_SDR_GPS #478: Wrong RX numeration in WSPR Module

MiBatu, 2021-02-26:
It seems that **RX0** in WSPR Extension module "Admin Interface" page is actually **RX1** for the rest of pages ("Status" page in Admin, "Users").

Please find screenshots below:
[image]
[image]

KiwiSDR Config: v1.436, 8 SDR channels

jks-prv, 2021-02-26:
No, it is fine. In "Autorun N" on the WSPR extensions admin page, "N" does not refer to an rxN channel number. It's just an indication of how many autorun processes are available. So 8 (0 - 7) in the case of a Kiwi configured for 8 channels.

The autorun processes are all started together when the Kiwi server is started. They all compete for rxN channel assignment just like simultaneous regular connections would. But since this process is done at server startup all the channels are usually empty and you almost always see autorun0 = rx0, autorun1 = rx1 etc.

BUT, a Kiwi in 8-channel mode is a special exception. That mode has waterfall on the first two channels, rx0 and rx1. The rest get only a less-capable audio FFT. In that mode you want the WSPR autorun to begin allocating using rx2 - rx7 and leave rx0/1 for last so regular user connections are given priority to rx0/1. That is exactly what has happened in your second image. You have enabled 7 autorun processes. 80m to 17m were assigned rx2 - 7 and 15m assigned rx0. rx1 with waterfall was then available for your user connection later on.

jks-prv, 2021-02-26:
This situation is documented directly in the text on the "mode" tab of the admin page where the Kiwi channel configuration is setup.

MiBatu, 2021-02-26:
Thank you, John for the detailed answer. Perhaps, this is just user first impression (a bit unexpected mismatch) and I understand the logic you described.

closed 2021-02-27 by jks-prv