jks-prv/Beagle_SDR_GPS #498: log timestamping - inconsistent timezones

csylvain, 2021-03-23:
the timestamps in the system log would start off in my local timezone (determined by lat:lon available to kiwisdr daemon?) and then slew to UTC after GPS acquisition on every reboot.

account 'root' and 'debian' on beagleboard report UTC timezone from command line.

during boot up, these lines appear in log:
```
Tue Mar 23 14:33:02 00:00:39.314 ....      TIMEZONE: lat/lon from admin public config: (39.599998, -78.900002)
Tue Mar 23 14:33:03 00:00:40.300 ....      IPINFO: public ip 98.219.234.52 from get.geojs.io
Tue Mar 23 14:33:03 00:00:40.303 ....      IPINFO: lat/lon = (39.649399, -78.930603) from get.geojs.io
Tue Mar 23 14:33:03 00:00:40.307 ....      task pub_NET:P2:T003(-) exited by returning
Tue Mar 23 14:33:04 00:00:41.242 ....      TIMEZONE: from timezonedb.com for (39.599998, -78.900002): utc_offset=-14400/-4.0 dst_offset=0/0.0
Tue Mar 23 14:33:04 00:00:41.248 ....      TIMEZONE: "EDT", "America\/New_York"
Tue Mar 23 14:33:04 00:00:41.250 ....      task get_TZ:P2:T004(-) exited by returning
```

i checked hwclock to verify RTC is set to UTC -- confirmed.
yet: the timedatectl configuration (systemd) reported RTC *not* set to local time (UTC in this case).

i have changed the following
```
sudo timedatectl set-local-rtc 1

timedatectl
      Local time: Tue 2021-03-23 15:39:19 UTC
  Universal time: Tue 2021-03-23 15:39:19 UTC
        RTC time: Tue 2021-03-23 15:39:19
       Time zone: Etc/UTC (UTC, +0000)
 Network time on: yes
NTP synchronized: yes
 RTC in local TZ: yes

Warning: The system is configured to read the RTC time in the local time zone.
         This mode can not be fully supported. It will create various problems
         with time zone changes and daylight saving time adjustments. The RTC
         time is never updated, it relies on external facilities to maintain it.
         If at all possible, use RTC in UTC by calling
         'timedatectl set-local-rtc 0'.
```

now the entire system is in UTC *and* the RTC is configured for UTC local timezone.
upon reboot, the log now begins with a UTC timestamp (not my local time zone) and does no longer slew forward some many hours from my local time to UTC shortly after completing the system reboot and acquiring GPS.

jks-prv, 2021-03-23:
This is incorrect. The Kiwi code does **not** change the Linux/Debian timezone. It should always be UTC. As the Beagle does not have an RTC (real-time clock) driver currently I don't know what set-local-rtc does.

The Kiwi code does attempt to determine the timezone so it can show local time in the user interface in addition to UTC. It can only do this if it can determine the lat/lon. There are three ways of doing that: via GPS if it has a signal, from public ip address geo-location or, as a last resort, if the admin has entered a lat/lon on the admin page.

csylvain, 2021-03-23:
okay, then.  but i did describe the issue with the timestamps in the Log
tab of the Admin interface.  having made the change specified, the log
entry timestamps do not slew 4 hours ahead after the GPS is reacquired.  i
made no change to the timezone of any account.  only set a flag to notify
how to interpret output from hwclock.

csylvain, 2021-03-23:
timezone for UI is still:
>
> TIMEZONE: "EDT", "America\/New_York"
>
>

csylvain, 2021-03-23:
here is another way to describe the issue:
the initial timestamps used in the log file are the local time and not UTC
when set-local-rtc is not set. the systemd is getting time from ntpdate?
and then setting the time in the kernel (which can be obtained via
/sbin/hwclock ) to the local time zone.  in only minutes later, once time
is obtained from GPS, then the log timestamps are clearly getting slewed
forward as many hours as it takes to catch back up with UTC.

does this help?

jks-prv, 2021-03-24:
What are you talking about? Your log attachment does not show the UTC timestamps (first column) "slewing ahead" when the TZ is acquired. They are always in UTC. The only possible issue is if you deliberately set the Linux TZ instead of leaving it UTC as shipped in the image from the factory. People who mess with Linux in this way get what they deserve.