jks-prv/Beagle_SDR_GPS #511: Feature Request: Hide certain IP From log

TGCFabian, 2021-04-09:
Hi! I love the KiwiSDR, and run a FT8Skimmer on it, there is just one slightly annoying side effect

This FT8Skimmer runs on the KiwiSDR Itself, which generates ALOT of connections through out a day, my request is:

Hide a certain IP(Preferably 127.0.0.1, or more, configurable through admin panel or file on the kiwi) from the logs,
This will help to keep the log files on my KiwiSDR Readable and clean, and most likely for some others too,
This is not a NEEDED function/feature, but one I would highly appreciate!

And a big thanks to all the developers for the awesome work on the KiwiSDR So far!

jks-prv, 2021-04-09:
How about this suggestion, because it would be much easier for me to implement. An admin switch that says "don't log connections from local ip address". I already have code to do local address detection (i.e. loopback 127.0.0.1, 10.x.x.x, 172.16-32.x.x, 192.168.x.x). Would that be sufficient?

Are the only two log messages you would want to hide the "arrived" and "leaving" ones? Are there any others?
```
Apr  9 04:24:02 ai kiwid: 00:02:04.921 0... 0     7550.00 kHz  am z0  "kiwirecorder.py" 127.0.0.1 (ARRIVED)
Apr  9 04:24:13 ai kiwid: 00:02:15.326 .... 0     7550.00 kHz  am z0  "kiwirecorder.py" 127.0.0.1 (LEAVING after 0:00:11)
```

TGCFabian, 2021-04-09:
Spot on! That would indeed be an easier implementation!

And yup! Exactly only those 2!

Thanks for the awesome fast reply! Appriciate it!!

jks-prv, 2021-04-09:
Okay, should be in the next release.

jks-prv, 2021-04-13:
in v1.451

closed 2021-04-13 by jks-prv