jks-prv/Beagle_SDR_GPS #576: Expanding KiwiSDR Software to Other Hardware

bhall7, 2022-01-21:
Is there any possibility of potentially expanding the KiwiSDR server software support other hardware such as Raspberry Pi and RTL-SDR or Airspy radios? I love the KiwiSDR web interface, especially on mobile. Unfortunately, I have not yet found a similar web interface that has the ease of use as KiwiSDR.

G8JNJ, 2022-01-22:
The KiWi software was forked from OpenWebRX https://www.openwebrx.de/ .

OpenWebRX is not as developed as the KiWi GUI and its newer features, but it will support Raspberry Pi , RTL-SDR and SDR Play.

There is a support forum at https://groups.io/g/openwebrx

bhall7, 2022-01-22:
Yes, I’m aware of OpenWebRX, and the fact that KiwiSDR was forked from it years ago. Unfortunately, it doesn’t have the features that KiwiSDR has—most notably, the ability to pan the waterfall left/right (set center frequency on the fly via drag and drop in the UI).

I understand that it is for KiwiSDR hardware, first and foremost. I was just wondering, and wishing, if at some point in the future KiwiSDR would ever possibly support other hardware like Pi + RTL-SDR/Airspy hardware. It would be so amazing.