jks-prv/Beagle_SDR_GPS #889: Decoder for DCF77 does not acquire sync

hochwasser, 2024-05-04:
The decoder for DCF77 does not acquire a sync even with very good reception.
(like on http://21283.proxy.kiwisdr.com:8073/)
I don't understand how the synchronization works, so I can't help you more :-(

jks-prv, 2024-05-08:
Hi. Yes, many of the decoders work, but they have an initial phase setup problem. The DCF77 decoder is especially bad. You can get it to sync by pushing the yellow "re-sync" button until the red and black signal traces both are in phase (i.e. move up and down together instead of opposite to each other). Sometimes it takes many button pushes until this happens. I need to create a software fix for this problem.

The MSF decoder is opposite. The red & black traces need to be out-of-phase. And it usually starts up this way. But not always.

jks-prv, 2024-05-10:
Okay, should be fixed for any Kiwi running the latest release (v1.679 or later).

closed 2024-05-10 by jks-prv